Serial-Studio/Serial-Studio #267: ESP32的重启会使Serial Stuido 程序崩溃

Tinyu-Zhao, 2025-02-19:
### **对于中国用户 / For Chinese users, please delete this part if you speak English**

### **1. 描述问题**

如果通过Serial Stuido 连接到ESP32的串口, 那么操作ESP32重启时会造成Serial Stuido程序崩溃

### **2. 截图（如果适用）**

如果通过Serial Stuido 连接到ESP32串口, 并按下重启键, Serial Studio程序崩溃
https://github.com/user-attachments/assets/e638d251-42c1-41b2-a96e-22de51e2ab9a

通过picocom 连接到ESP32串口, 并按下重启键,输出正常
https://github.com/user-attachments/assets/c936cb8c-e2a2-4d9a-877c-6ee02e6ef9f3

### **3. 上传您的 JSON 地图文件**

我不知道从哪里获取JSON 地图文件

### **4. 提供一个最小的 Arduino 程序以复现问题**

> 提供一个最小的 Arduino 程序有助于我复现导致问题的环境和设置。  

无需程序

alex-spataru, 2025-03-04:
Thanks! I’ll investigate the issue and let you know if I identify a probable cause. Could you confirm if this still occurs with the latest continuous release?

---

谢谢！我会调查这个问题，并在找到可能的原因后通知您。请确认此问题是否仍然在最新的持续发布版本中发生？

alex-spataru, 2025-03-04:
By the way, I noticed you’re using “Quick Plot” mode, which might be causing the crash. This happens because the ESP32 reboot log is displayed in the terminal, and Serial Studio attempts to parse the incoming data as an array of numbers by splitting it using commas (`,`).

To avoid this issue, you might try:
- Sending only the sensor value (e.g., "0" instead of "Lux: 0").
- Using Project Mode to plot the data. In this case, you may need a key/value frame parser function similar to the one described [here](https://github.com/Serial-Studio/Serial-Studio/wiki/Data-Flow-in-Serial-Studio#example-parsing-key-value-pairs-with-a-user-defined-keymap).

If you’re not interested in plotting data, setting the operation mode to “No Parsing” might be a better option. You can read more about operation modes [here](https://github.com/Serial-Studio/Serial-Studio/wiki/Operation-Modes).

---

顺便说一下，我注意到您正在使用“Quick Plot”模式，这可能是导致崩溃的原因。这是因为 ESP32 重新启动时会在终端显示日志，而 Serial Studio 会尝试使用逗号（`,`）分隔传入数据并解析为数字数组。

为避免此问题，您可以尝试：
	•	仅发送传感器数值（例如 "0"，而不是 "Lux: 0"）。
	•	使用 Project Mode 来绘制数据。在这种情况下，您可能需要使用类似于[此处](https://github.com/Serial-Studio/Serial-Studio/wiki/Data-Flow-in-Serial-Studio#example-parsing-key-value-pairs-with-a-user-defined-keymap)描述的 键/值帧解析函数。

如果您不需要绘制数据，将操作模式设置为 “No Parsing” 可能是更好的选择。您可以在[这里](https://github.com/Serial-Studio/Serial-Studio/wiki/Operation-Modes)了解更多关于操作模式的信息。

Tinyu-Zhao, 2025-03-04:
> By the way, I noticed you’re using “Quick Plot” mode, which might be causing the crash. This happens because the ESP32 reboot log is displayed in the terminal, and Serial Studio attempts to parse the incoming data as an array of numbers by splitting it using commas (`,`).顺便说一句，我注意到你正在使用“快速绘图”模式，这可能会导致崩溃。发生这种情况的原因是 ESP32 重启日志显示在终端中，而 Serial Studio 尝试通过使用逗号 ( `,` ) 将其拆分，将传入数据解析为数字数组。
> 
> To avoid this issue, you might try:为了避免此问题，您可以尝试：
> 
> * Sending only the sensor value (e.g., "0" instead of "Lux: 0").仅发送传感器值（例如，“0”而不是“Lux: 0”）。
> * Using Project Mode to plot the data. In this case, you may need a key/value frame parser function similar to the one described [here](https://github.com/Serial-Studio/Serial-Studio/wiki/Data-Flow-in-Serial-Studio#example-parsing-key-value-pairs-with-a-user-defined-keymap).使用项目模式绘制数据。在这种情况下，您可能需要一个类似于[此处](https://github.com/Serial-Studio/Serial-Studio/wiki/Data-Flow-in-Serial-Studio#example-parsing-key-value-pairs-with-a-user-defined-keymap)描述的键/值框架解析器函数。
> 
> If you’re not interested in plotting data, setting the operation mode to “No Parsing” might be a better option. You can read more about operation modes [here](https://github.com/Serial-Studio/Serial-Studio/wiki/Operation-Modes).如果您对绘制数据不感兴趣，将操作模式设置为“无解析”可能是更好的选择。您可以[在此处](https://github.com/Serial-Studio/Serial-Studio/wiki/Operation-Modes)阅读有关操作模式的更多信息。
> 
> 顺便说一下，我注意到您正在使用“Quick Plot”模式，这可能是导致崩溃的原因。这是因为 ESP32 重新启动时会在终端显示日志，而 Serial Studio 会尝试使用逗号（`,`）分隔传入数据并解析为数字数组。
> 
> 为避免此问题，您可以尝试： • 仅发送传感器数值（例如 "0"，而不是 "Lux: 0"）。 • 使用 Project Mode 来绘制数据。在这种情况下，您可能需要使用类似于[此处](https://github.com/Serial-Studio/Serial-Studio/wiki/Data-Flow-in-Serial-Studio#example-parsing-key-value-pairs-with-a-user-defined-keymap)描述的 键/值帧解析函数。
> 
> 如果您不需要绘制数据，将操作模式设置为 “No Parsing” 可能是更好的选择。您可以在[这里](https://github.com/Serial-Studio/Serial-Studio/wiki/Operation-Modes)了解更多关于操作模式的信息。

谢谢你的帮助, 我使用的是v3.0.6版本, 经过测试, 在快速绘图模式下 ESP32的重启会造成Serial Stuido崩溃, 在No Parsing 模式下, 可以正常工作.