Serial-Studio/Serial-Studio #272: Crash when trying to open device

FlyingRatBull, 2025-02-28:
### **1. Describe the bug**

When connecting to a device with _Quick Plot_ frame parsing, the app just crashes and closes.

### **2. Screenshots (if applicable)**

[image]

### **3. Upload your JSON map file**
Using _Quick Plot_

### **4. Provide a minimal Arduino program to replicate the issue**

```cpp
void setup() {
    Serial.begin(9600);
    while (!Serial);
}

void loop() {
    Serial.println("21.63,250,0.0,0,0.0,0,0.0,0");
    delay(50);
}
```

### Environment
* Ubuntu 22.04 LTS
* Tested Serial-Studio-*-Linux-x86_64.AppImage Versions: 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.0.4, 3.0.5, 3.0.6

### Coredump

[core.zip](https://github.com/user-attachments/files/19023889/core.zip)

FlyingRatBull, 2025-02-28:
Just tested it with the latest Continuous Build (Feb 4 2025) and get a different bug:

When selecting the device, I get the error
`qrc:/qml/MainWindow/Panes/Toolbar.qml:327: ReferenceError: Cpp_MQTT_Client is not defined`
And the _Connect_ Button is still disabled.

alex-spataru, 2025-03-04:
Hi! Please let me know if this issue persists with the latest continuous build. I'll take a look at the core dump file when I have the chance. Thanks!

FlyingRatBull, 2025-03-04:
This seems to have fixed the problem, thanks!

closed 2025-04-21 by alex-spataru