Serial-Studio/Serial-Studio #275: Missing UI elements on continuous version

TheAlbertDev, 2025-03-16:
**Bug Report Template**  

---

### **1. Describe the bug**  

> Provide a clear and concise description of the bug you’re experiencing.  

While testing the functionality that allows configuring what is displayed on the X-axis of the graphs (#218 ), that option does not appear in the latest version (continuous), along with other icons/buttons such as the MQTT one.  

### **2. Screenshots (if applicable)**  

> Add screenshots to help illustrate the problem, if applicable.  

[image]

[image]

[image]

### **3. Upload your JSON map file**  

> This is essential for reproducing any UI-related issues.  

N/A  

### **4. Provide a minimal Arduino program to replicate the issue**  

> Sharing a minimal Arduino sketch helps me replicate the environment and setup that causes the bug.  

N/A  

### **5. Important Notes**  

- Non-critical bugs will be treated with **lower priority than critical bugs**.  
- Donations are always welcome to help keep the project alive and ensure a timely response. You can support the project here: [Donate Now](https://www.paypal.com/donate?hosted_button_id=XN68J47QJKYDE).  
- **I am an open-source developer.** I provide this software as-is and cannot be responsible for long-term maintenance or personalized service. The source code is available freely, and if the issue is urgent, you’re encouraged to fix it yourself or hire someone (including myself) to do so.  

Thank you for understanding and for helping improve the software!

alex-spataru, 2025-03-28:
Hi! 

I’m transitioning the project to a dual-license model: GPLv3 for the basic (free) version and a paid version with extended features. This change is partly due to discussion #248 and also because some Qt modules I want to use (like MQTT—not the current qmqtt dependency, or Qt Graphs) are only available under GPLv3 or a commercial license.

All of the code will of course remain open-source (now under GPL3). But to support advanced features, I need to implement some form of licensing/activation for the commercial version. So yeah, the continuous build is kind of "unintegrated" right now while I figure out how to best approach this and carve out time to work on it.

My goal is balance: I don’t want to cripple the free version, but I do want to offer enough value in the paid version for users to justify supporting the project financially. Features like proper MQTT support, CANBus/ModBus integration, and XY plotting are good candidates for the commercial tier.

If I could turn Serial Studio into a primary income stream, I’d happily pour full-time effort into making it even better.

Thanks for the patience and support.

TheAlbertDev, 2025-04-10:
> Hi!
> 
> I’m transitioning the project to a dual-license model: GPLv3 for the basic (free) version and a paid version with extended features. This change is partly due to discussion [#248](https://github.com/Serial-Studio/Serial-Studio/discussions/248) and also because some Qt modules I want to use (like MQTT—not the qmqtt dependency in stable release, or Qt Graphs) are only available under GPLv3 or a commercial license.
> 
> All the code will stay open-source under GPLv3. That said, I still need to figure out how to handle licensing or activation for the commercial features—even with the full source available. Once I figure that, it could be a useful model for other devs trying to do the same. For now, the continuous build is a bit messy while I work through this and find time to put it all together properly.
> 
> My goal is balance, I don’t want to cripple the free version, but I do want to offer enough value in the paid version for users to justify supporting the project financially. Features like proper MQTT support, CANBus/ModBus integration, and XY plotting are good candidates for the commercial tier.
> 
> If I could turn Serial Studio into a primary income stream, I’d happily pour full-time effort into making it even better.
> 
> Thanks for the patience and support.

Totally understandable. Thanks for all your work.

closed 2025-04-10 by TheAlbertDev