Serial-Studio/Serial-Studio #285: Filtering out duplicate lines

tyeth, 2025-05-01:
Hey, I have a stream of log data, and some of it's very noisy and repetitive, I'd like to be able to filter out duplicate lines from the console, also having some kind of highlighting of lines in the console based on a filter / regex etc would be great for my needs.

I'm sure there was a good tool for that already so maybe I just need to find it again, but I also like this tool a lot and wondered if I could bend it to my will, and what the best way of doing that would be...

avionicharshit-byte, 2025-11-25:
I would like to work on this issue. I'm planning to implement:
  1. A checkbox to toggle duplicate line filtering
  2. Logic to detect and skip consecutive duplicate lines

  My approach would be to modify the Console::append() method to track the last displayed line and skip duplicates when filtering is enabled. Does this sound good? Is anyone else already working on this? @alex-spataru @tyeth

avionicharshit-byte, 2025-12-03:
any updates ? @tyeth @alex-spataru