Serial-Studio/Serial-Studio #286: [Feature] Submit to flathub

JamesBrosy, 2025-05-06:
Hi! I have submitted this project to [flathub](https://github.com/flathub/flathub/pull/6460). I have [built](https://docs.flathub.org/docs/for-app-authors/submission/#before-submission) and tested the submission locally. There are no problems so far.

alex-spataru, 2025-05-07:
Hi!

Thanks a lot for your contribution and for pushing this to Flathub, really appreciate it!

Let me know if you hit any roadblocks or need anything from my side to keep things moving. I don’t have time to test across all Linux distros (hence sticking to AppImages), so your effort here is seriously valuable.

**Quick heads-up for the next release:** the project will be dual-licensed under GPL3 and commercially. GitHub binaries will be the commercial build, which includes extra modules (like 3D plots and possibly CAN Bus/Modbus support). These extra features are activated via a license key and **not** included in the GPL3 binary to stay compliant, even though all the code is open source. Most of these features rely on Qt libraries/plugins that are only available with a commercial Qt license.

I believe most Flathub users would prefer to use the GPL3-only version...meaning the build must use the open source version of Qt, e.g. the version you get through package managers (like `apt-get install qt*`).  The `CMakeLists.txt` is set up to detect and build accordingly (open source vs. commercial) based on available Qt modules, no flags or manual changes needed (from my testing; feedback on master branch or continuous build is welcome).

If you see any way to improve CMake for open source packaging, let me know, I think it's a good window to sort out any issues.

Thanks again for your work on this!

JamesBrosy, 2025-05-07:
Thanks for your reply. I have successfully built this software through Flatpak and it works well. The [AppStream MetaInfo file](https://github.com/JamesBrosy/flathub/blob/io.github.serial_studio.Serial_Studio/io.github.serial_studio.Serial_Studio.metainfo.xml) could be improved, Please check out it and replenish in time. I would like to upstream it to this project if possible: 

For example, add `io.github.serial_studio.Serial_Studio.metainfo.xml` to `app/deploy/linux` and add the code below to CMakeLists.txt

```diff
elseif(UNIX)
  install(TARGETS ${PROJECT_EXECUTABLE} RUNTIME DESTINATION /usr/bin)

  install(
    FILES ${CMAKE_CURRENT_SOURCE_DIR}/deploy/linux/serial-studio.svg
    DESTINATION /usr/share/pixmaps
  )

+ install(
+   FILES ${CMAKE_CURRENT_SOURCE_DIR}/deploy/linux/io.github.serial_studio.Serial_Studio.metainfo.xml
+   DESTINATION /usr/share/metainfo
+ )
+
install(
  FILES ${CMAKE_CURRENT_SOURCE_DIR}/deploy/linux/serial-studio.desktop
  DESTINATION /usr/share/applications
)
```

alex-spataru, 2025-06-09:
Hi @JamesBrosy!

Just a quick update, I’ve created a [new release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/v3.1.0) that includes all the latest features. I also updated the [`io.github.serial_studio.Serial_Studio.metainfo.xml`](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/deploy/linux/io.github.serial_studio.Serial_Studio.metainfo.xml) file to better match the README, added the 3.1.0 release links, and swapped in the screenshot from the main README.

Let me know if you need anything on the Flathub publishing side. It took some time, but this release packs in a lot of useful improvements.

Thanks!

alex-spataru, 2025-06-11:
Hi! Thanks again for your help. Just a quick update, I’ve created a new release. The only change I made to the [serial-studio.metainfo.xml](https://github.com/Serial-Studio/Serial-Studio/blob/master/app/deploy/linux/serial-studio.metainfo.xml) file was updating the version to 3.1.2, which is now the latest release.

Really appreciate your support!

alex-spataru, 2025-06-11:
Another question...I saw in your screenshot in the PR the "freemium" version of Serial Studio, is that screenshot from the flatpak build or the official binary? 

If it's from the flatpak build please let me know, it might be a bug in my code...open source users should see "A you're using the GPL3 version of Serial Studio" message in the console, not the freemium message.

JamesBrosy, 2025-06-12:
> Another question...I saw in your screenshot in the PR the "freemium" version of Serial Studio, is that screenshot from the flatpak build or the official binary?
>
> If it's from the flatpak build [...]

Yes, it is. The flatpak version of this software is built from the source code, so It's the same software you released on github If there are any issues with the license, you can supplement project license in the metainfo.xml. There is no mandatory requirement for software published on Flathub to be completely open source.

JamesBrosy, 2025-06-12:
> **Quick question:** is there a way to run tests using the continuous release or master branch? I was thinking of pushing these changes directly to the `v3.1.2` tag, but I’m not sure if the CI script used by FlatHub would pick that up.
> 

If the software is released to Flathub, every time there is a new version, and Flathub automatically generates a PR, then tests it, and builds it, like this https:github.com/flathub/com.google.Chrome/pull/401.

alex-spataru, 2025-06-14:
> Yes, it is. The flatpak version of this software is built from the source code, so It's the same software you released on github. If there are any issues with the license, you can supplement project license in the metainfo.xml. There is no mandatory requirement for software published on Flathub to be completely open source.  
> For example, Discord is a non-free software, and project license is like [this](https://github.com/flathub/com.discordapp.Discord/blob/master/com.discordapp.Discord.appdata.xml#L10).  
> And you can also add a link, like [this](https://github.com/flathub/com.tencent.WeChat/blob/master/com.tencent.WeChat.metainfo.xml#L6).

Thanks, that’s very helpful. I’ve updated the `metainfo.xml` file to indicate a proprietary license and added a link to the full license text.

I think offering a GPLv3 version on Flathub could make sense for some users, but realistically, most just want Serial Studio to “just work.” From that perspective, the proprietary binary makes more sense.

That said, using Flathub will also help mitigate the current limitations we have with `arm64` builds. Our pipeline currently requires GLibC ≥ 2.38, which isn’t yet standard across all distros...Flatpak helps bypass that problem with portable, sandboxed builds.

> If the software is released to Flathub, every time there is a new version, and Flathub automatically generates a PR, then tests it, and builds it, like this https://github.com/flathub/com.google.Chrome/pull/401.

Perfect! That kind of automated flow is exactly what we need.

I just tagged a new release that includes a validated `metainfo.xml`, so the PR should go through more smoothly. Let me know if you run into any issues.

Also, shoot me an email at [alex@serial-studio.com](mailto:alex@serial-studio.com) once Serial Studio goes live on Flathub. I think you deserve a Pro license key for the help:  [VALUE-FOR-LICENSE.md](https://github.com/Serial-Studio/Serial-Studio/blob/master/VALUE-FOR-LICENSE.md)

Thanks again!

JamesBrosy, 2025-06-15:
> Also, shoot me an email at [alex@serial-studio.com](mailto:alex@serial-studio.com) once Serial Studio goes live on Flathub. I think you deserve a Pro license key for the help: [VALUE-FOR-LICENSE.md](https://github.com/Serial-Studio/Serial-Studio/blob/master/VALUE-FOR-LICENSE.md)另外，Serial Studio 在 Flathub 上线后，请给我发送电子邮件至 [alex@serial-studio.com](mailto:alex@serial-studio.com) 。我认为你应该获得一个专业版许可证密钥作为对你帮助的回报： [VALUE-FOR-LICENSE.md](https://github.com/Serial-Studio/Serial-Studio/blob/master/VALUE-FOR-LICENSE.md)
> 

Thank you very much for your gift, and it helps me a lot. As a Linux user with a system cleanliness obsession, I don't want to manage a bunch of software data and configuration files, so I choose `flatpak`. Then I start actively packaging software that I commonly use on Linux but don't have on Flathub.

alex-spataru, 2025-06-16:
@JamesBrosy Serial Studio is now available on [Flathub](https://flathub.org/apps/com.serial_studio.Serial-Studio)! Thanks again for your help. When you get a chance, please send me an email with the address you’d like to use to manage your Serial Studio license.

closed 2025-06-16 by alex-spataru

JamesBrosy, 2025-06-17:
> [@JamesBrosy](https://github.com/JamesBrosy) Serial Studio is now available on [Flathub](https://flathub.org/apps/com.serial_studio.Serial-Studio)! Thanks again for your help. When you get a chance, please send me an email with the address you’d like to use to manage your Serial Studio license.

Hello, I've sent you an email, please check it.

alex-spataru, 2025-06-17:
@JamesBrosy Just sent you an email with the license key, thanks for all the help!