Serial-Studio/Serial-Studio #289: Crash after project save

maximeg, 2025-06-02:
Hey,

I've been using SerialStudio, that is a nice piece of software, thanks for it.

I've been experiencing systematic crashes when I save a project. The project ends up being saved though.

I use v3.0.6 with MacOS 15.5, ARM-64.

```
Process:               Serial-Studio [29809]
Path:                  /Applications/Serial Studio.app/Contents/MacOS/Serial-Studio
Identifier:            org.alexspataru.serial-studio
Version:               3.0.6 (???)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-06-02 18:34:21.2134 +0200
OS Version:            macOS 15.5 (24F74)
Report Version:        12
Anonymous UUID:        B97EB376-DC46-E9CD-EE1B-BDCF979ECE26

Time Awake Since Boot: 1100000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000048
Exception Codes:       0x0000000000000001, 0x0000000000000048

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [29809]

VM Region Info: 0x48 is not in any region.  Bytes before following region: 4377198520
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104e6c000-10506c000    [ 2048K] r-x/r-x SM=COW  /Applications/Serial Studio.app/Contents/MacOS/Serial-Studio

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   Serial-Studio                 	       0x104eefcb4 JSON::Group::datasetCount() const + 0
1   Serial-Studio                 	       0x104eb0914 Widgets::Accelerometer::updateData() + 104
2   QtCore                        	       0x107ad996c 0x1079f8000 + 924012
3   Serial-Studio                 	       0x104ea35ac UI::Dashboard::resetData(bool) + 680
4   QtCore                        	       0x107ad996c 0x1079f8000 + 924012
5   Serial-Studio                 	       0x104eea738 JSON::FrameBuilder::loadJsonMap(QString const&) + 84
6   Serial-Studio                 	       0x104ed4178 JSON::ProjectModel::saveJsonFile() + 2700
7   Serial-Studio                 	       0x104e83b80 0x104e6c000 + 97152
8   Serial-Studio                 	       0x104e799cc JSON::ProjectModel::qt_metacall(QMetaObject::Call, int, void**) + 188
9   QtQml                         	       0x108515578 0x1083ec000 + 1217912
10  QtQml                         	       0x108513660 QV4::QObjectMethod::callPrecise(QQmlObjectOrGadget const&, QQmlPropertyData const&, QV4::ExecutionEngine*, QV4::CallData*, QMetaObject::Call) + 492
11  QtQml                         	       0x108518fe8 QV4::QObjectMethod::callInternal(QV4::Value const*, QV4::Value const*, int) const + 1884
12  QtQml                         	       0x1085871a8 0x1083ec000 + 1683880
13  QtQml                         	       0x108586010 0x1083ec000 + 1679376
14  QtQml                         	       0x1084cbe30 0x1083ec000 + 917040
15  QtQml                         	       0x1084cb5cc QV4::Function::call(QObject*, void**, QMetaType const*, int, QV4::ExecutionContext*) + 532
16  QtQml                         	       0x10861f3b0 QQmlJavaScriptExpression::evaluate(void**, QMetaType const*, int) + 280
17  QtQml                         	       0x1085ccd28 QQmlBoundSignalExpression::evaluate(void**) + 756
18  QtQml                         	       0x1085cd2ac 0x1083ec000 + 1970860
19  QtQml                         	       0x1086454e8 QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) + 496
20  QtCore                        	       0x107ad9490 0x1079f8000 + 922768
21  QtQml                         	       0x1086e4df0 QQmlVMEMetaObject::metaCall(QObject*, QMetaObject::Call, int, void**) + 4452
22  QtQml                         	       0x108514178 0x1083ec000 + 1212792
23  QtQml                         	       0x108513660 QV4::QObjectMethod::callPrecise(QQmlObjectOrGadget const&, QQmlPropertyData const&, QV4::ExecutionEngine*, QV4::CallData*, QMetaObject::Call) + 492
24  QtQml                         	       0x108518fe8 QV4::QObjectMethod::callInternal(QV4::Value const*, QV4::Value const*, int) const + 1884
25  QtQml                         	       0x1085871a8 0x1083ec000 + 1683880
26  QtQml                         	       0x108586010 0x1083ec000 + 1679376
27  QtQml                         	       0x1084cbe30 0x1083ec000 + 917040
28  QtQml                         	       0x1084cb5cc QV4::Function::call(QObject*, void**, QMetaType const*, int, QV4::ExecutionContext*) + 532
29  QtQml                         	       0x10861f3b0 QQmlJavaScriptExpression::evaluate(void**, QMetaType const*, int) + 280
30  QtQml                         	       0x1085ccd28 QQmlBoundSignalExpression::evaluate(void**) + 756
31  QtQml                         	       0x1085cd2ac 0x1083ec000 + 1970860
32  QtQml                         	       0x1086454e8 QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) + 496
33  QtCore                        	       0x107ad9490 0x1079f8000 + 922768
34  QtQuick                       	       0x1066de4a8 QQuickMouseArea::setPressed(Qt::MouseButton, bool, Qt::MouseEventSource) + 724
35  QtQuick                       	       0x1066ded14 QQuickMouseArea::mouseReleaseEvent(QMouseEvent*) + 152
36  QtQuick                       	       0x1066bf94c QQuickItem::event(QEvent*) + 780
37  QtWidgets                     	       0x105b34760 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 336
38  QtWidgets                     	       0x105b3571c QApplication::notify(QObject*, QEvent*) + 504
39  QtCore                        	       0x107a89584 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
40  QtQuick                       	       0x10684e9d8 QQuickDeliveryAgentPrivate::deliverMatchingPointsToItem(QQuickItem*, bool, QPointerEvent*, bool) + 472
41  QtQuick                       	       0x10684df84 QQuickDeliveryAgentPrivate::deliverUpdatedPoints(QPointerEvent*) + 248
42  QtQuick                       	       0x106849710 QQuickDeliveryAgentPrivate::deliverPointerEvent(QPointerEvent*) + 2244
43  QtQuick                       	       0x106847960 QQuickDeliveryAgentPrivate::handleMouseEvent(QMouseEvent*) + 1316
44  QtQuick                       	       0x1068470a0 QQuickDeliveryAgent::event(QEvent*) + 112
45  QtQuick                       	       0x10675e5f0 QQuickWindow::event(QEvent*) + 3380
46  QtQuick                       	       0x10676d554 QQuickWindowQmlImpl::event(QEvent*) + 460
47  QtWidgets                     	       0x105b34760 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 336
48  QtWidgets                     	       0x105b3571c QApplication::notify(QObject*, QEvent*) + 504
49  QtCore                        	       0x107a89584 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
50  QtGui                         	       0x108b7b9b8 QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 1892
51  QtGui                         	       0x108bde194 QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 408
52  libqcocoa.dylib               	       0x1082033a4 0x1081e8000 + 111524
53  CoreFoundation                	       0x1962a5cd4 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 28
54  CoreFoundation                	       0x1962a5c68 __CFRunLoopDoSource0 + 172
55  CoreFoundation                	       0x1962a59d4 __CFRunLoopDoSources0 + 232
56  CoreFoundation                	       0x1962a4628 __CFRunLoopRun + 840
57  CoreFoundation                	       0x1962a3c58 CFRunLoopRunSpecific + 572
58  HIToolbox                     	       0x1a1d3827c RunCurrentEventLoopInMode + 324
59  HIToolbox                     	       0x1a1d3b31c ReceiveNextEventCommon + 216
60  HIToolbox                     	       0x1a1ec6484 _BlockUntilNextEventMatchingListInModeWithFilter + 76
61  AppKit                        	       0x19a1cbab4 _DPSNextEvent + 684
62  AppKit                        	       0x19ab6a5b0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
63  AppKit                        	       0x19a1bec64 -[NSApplication run] + 480
64  libqcocoa.dylib               	       0x108200d8c 0x1081e8000 + 101772
65  QtCore                        	       0x107a946c8 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 1112
66  QtCore                        	       0x107a89da8 QCoreApplication::exec() + 136
67  Serial-Studio                 	       0x104efd48c main + 1212
68  dyld                          	       0x195e1ab98 start + 6076
```

alex-spataru, 2025-06-02:
Hi, and thanks for the bug report. Could you share the project file that’s causing the application to crash? I’d like to run it through a debugger to identify exactly what’s triggering the issue.

Appreciate your help.

alex-spataru, 2025-06-02:
I’ve just pushed a commit that should fix the issue and the compass issue (#291). You can test it by installing the latest [continuous release](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous). Note that this build may have some bugs, as I’m in the middle of a dashboard UI redesign...but the main goal is to confirm whether this specific issue is resolved.

maximeg, 2025-06-03:
Hey, thanks for your reactivity!

I tested it and the save issue is fixed.
I couldn't test the Compass #291 bug, because all sub widgets are not showing. I've tried recreating without success in basic container or the other types.
If you do a re-work, consider adding reordering as well.
A widget like the Gyro, but driven by roll/yaw/pitch directly rather than gyro rates would be great as well!

maximeg, 2025-06-04:
Ah, I didn't see that. So the Compass is fixed!

For the reordering in Project editor, a generic widget/subwidget reordering would be neat and allow use of autolayout with a bit of custom, it would also allow to reorder items in "Data Grid" ....

I found a new bug 😅 : on re-connect, every data is doubled. on the Nth reconnect, there are N copies of the same data.
If the serial spits char by char, those chars are duplicated (ex: `hhheeellllllooo`). If it spits a full line, it is duplicated.
I can't tell if the real release is concerned or not. I happen to use a TTLtoUSB that was always plugged to my Mac and so could always be kept connected. Now that I spent my afternoon implementing USB CDC (thank you ST for shitty doc), my virtual serial port resets each time I run the microcontroller. So I get the bug now.

On a side note, this popup appears a bit violently, even if SerialStudio is not my current focus, and stays even if SerialStudio did auto-reconnect. Further more it keeps appearing and stacks if there are other disconnection. I would put something next to the Connect/Disconnect button on the top right. It would be more logical (if I don't have auto-reconnect, I would click to that button) and constrained inside SerialStudio.
[image]

maximeg, 2025-06-05:
While on the subject of console, if "Send DTR Signal" is checked, it is set to 1 on connect (asserted) but not set to 0 on disconnect (deasserted). It disturbs systems that rely on DTR to know that the serial port is indeed open to send data (due to USB control flow done by the host).

maximeg, 2025-06-05:
Hey)

Everything works!

I tested with the debugger to find out that the code from ST was not exposing "user land" the `wValue` from the `SET_CONTROL_LINE_STATE` `bRequest`...  I Well let's say today I learnt more about USB CDC that I originally wanted!

I'm new to STM32 too. I designed from scratch an ADAHRS board with a STM32U5, it works. Now I'm into the code part. I'm a bit surprised not to find a more "serious" thing codewise with ST libs/examples/doc compared to Rasp. Pi Pico for which I have a bit of experience (it is clumsy, but it is for amateur mostly). For USB, they are transitioning to (and want to shovel us) USBX and Azure RTOS. I used the legacy USB_Device, and despite of X years of usage, it feels like a 1 hour weekend project. None of their CDC examples / ST Forum answers / ... work without a serial console opened! Or they work, but they miss half of the messages if they simply abort transmission on `USBD_BUSY`! Weekend project level 😢 I understand why people just use UART to a dedicated USB TTL chip and don't mess with USB stack in STM32!

I realized also that the min/max feature of graphs (multiplot & individual graph) does not work sometimes. I 97% sure it was working when I tried 789a4ce. I didn't see `setRange` on the codebase for the axis so I couldn't see on you handle it (it is maybe with the `lowerseries`?).

Also for the new workspace thing, having the bottom bar to show them would be neat (instead of an access to the widget that we already see).

alex-spataru, 2025-06-08:
Hi,

Thanks for the follow-up. And yes, sounds like you went way deeper into USB CDC than anyone should have to just to get basic stuff working. I’m still pretty new to ST microcontrollers myself, so I feel your pain...right now I'm trying to get the NFC6 module to work. No surprise people just drop in a UART-to-USB chip and avoid the mess entirely.

On the Serial Studio side: the graph min/max issue was a regression I introduced while optimizing frame parsing...I was unintentionally overwriting all the datasets on every dashboard update, instead of just updating their values. That should be fixed in the latest continuous build. Please let me know if it’s still acting up.

Also, thanks for the suggestion on the workspace UI too. Right now I’m trying to replicate a more classic Windows-style layout to help new users get comfortable, especially since one client wants to use it as an HMI for industrial controls. Over time, I want to refine the design based on more feedback from the community. I also added window snapping and the ability to reorder windows in auto-layout mode. You can just drag one over another to switch their positions.

Thanks!

maximeg, 2025-06-09:
Hello, min/max works now. Thanks.