Serial-Studio/Serial-Studio #296: Attitude widget

maximeg, 2025-06-09:
Hey,

I create a ticket for this even if we discussed it :)

It would be nice to have a widget like the Gyroscope one but that would take directly Roll, Pitch and Yaw.
[image]

Indeed, I found that the Gyroscope widget directly performs integration from rates, but it is quite naive and lead to outstanding errors that are quite normal to have after direct integration without any data fusion (with a Kalman filter or else).

alex-spataru, 2025-06-10:
Hi! Added commits that fix both the gyroscope rendering issue and add a small toolbar button in the Gyroscope widget that allows users to toggle between enabling angle integration or not. As always, you can test on the [continuous](https://github.com/Serial-Studio/Serial-Studio/releases/tag/continuous) build to confirm if the issue was solved. Thanks for the bug report!

closed 2025-06-11 by alex-spataru

hellvesper, 2026-02-04:
Hi @alex-spataru , I'm confused with Gyroscope widget that shows attitude. I'm sending absolute values, my IMU performs internal fusing, and attitude widget prints actual value at the bottom "Pitch 30.10˚" but attitude shows ~ 60 deg, same if actual pitch 20˚ it shows ~40˚. 

[image]