Serial-Studio/Serial-Studio #304: 必须先发指令等有回复数据后才能切换到dashboard

dzx815, 2025-06-13:
[image]

如图所示，必须要发两次查询指令才能到这个仪表盘界面。

实际上我有一些操作按钮，但是在这个界面无法选中，只能手动输入指令交互到2次以上才能自动切换到dash board界面。

dzx815, 2025-06-14:
Can the operation buttons of the dashboard be directly displayed on the interface instead of having to click menu ->operation ->action every time in the secondary menu

可以把动作按钮直接平铺在仪表盘界面吗？每次操作都需要依次点击
菜单->操作->"对应的动作"才行

alex-spataru, 2025-06-14:
Hi! I'm currently investigating why the dashboard is only built after receiving the second data frame instead of the first. Once I identify the cause, I’ll push a fix and let you know.

> Can the operation buttons of the dashboard be displayed directly, instead of going through Menu → Operation → Action every time?

Yes. I’ve added a commit that shows a submenu with all available actions directly in the Dashboard:

[image]

You can enable or disable this behavior in the **Preferences** dialog.

---

你好！我正在调查为什么仪表盘要在接收到第二帧数据后才开始构建，而不是第一帧。一旦找到原因并修复，我会提交更新并通知你。

> 仪表盘的操作按钮能否直接显示，而不是每次都需要点击“菜单 → 操作 → 指令”？

可以。我已经提交了一个更新，现在所有可用的操作会作为子菜单直接显示在仪表盘界面中：

[image]

你可以在**首选项**对话框中启用或禁用这个功能。

alex-spataru, 2025-06-15:
> Hi! I'm currently investigating why the dashboard is only built after receiving the second data frame instead of the first. Once I identify the cause, I’ll push a fix and let you know.

Fixed. Drafted a new release containing the fixes. Thanks for the bug report!

---

> 嗨！我正在调查为什么仪表板要在接收到第二个数据帧后才开始构建，而不是第一个。一旦找到原因，我会修复它并通知你。

已修复。我已经创建了一个包含修复的新版本。感谢你的错误报告！

dzx815, 2025-06-16:
其实我想的是可以不发指令手动切换到仪表盘界面，
但是现在必须要发送收到指令才能切换到dashboard界面。

因为有一些操作按钮可以直接发送指令，而在控制台界面，我只能自己填写命令发送后才能切换到dashboard界面。

dzx815, 2025-06-16:
[image]

这里可以添加一个发送按钮吗？有的时候希望纯鼠标就能操作，比如是其他人给我发送的指令，我直接复制粘贴就可以发送了，而不用取操作键盘。
而且发送完成希望命令还在输入框，并不会主动清空。

Can we add a send button here? Sometimes I hope to be able to operate with just a mouse, for example, when someone sends me instructions, I can simply copy and paste them without having to access the keyboard.
And after sending, the command is still in the input box and will not be automatically cleared.

closed 2025-06-20 by dzx815