Serial-Studio/Serial-Studio #310: dashboard界面进入后是黑屏

dzx815, 2025-06-25:
[image]
如图所示，进入dashboard界面是黑屏状态

As shown in the figure, entering the dashboard interface is in a black screen state

alex-spataru, 2025-06-25:
Hi, I noticed this issue in the latest commits I made, the problem is the widget doesn’t render until a second frame triggers an update.

Are you still seeing this in the latest continuous build? The current fix isn’t perfect yet (it renders all table widgets, even minimized or closed ones) but it should resolve the immediate issue.

---

嗨，我在我最近的提交中注意到了这个问题，问题是这个小部件在接收到第二帧之前不会渲染，从而需要触发一次更新。

你在最新的持续构建中还遇到这个问题吗？目前的修复方案还不够理想（它会渲染所有表格小部件，包括已最小化或关闭的），但应该能解决这个即时问题。

dzx815, 2025-06-25:
我的数据帧是持续在发的（也就是一直在刷新），但是这个表格小部件始终为黑色，但是如果将这个部件设置为全屏，再切为小屏就正常了。

My data frames are sent continuously (that is, they are always refreshed), but this table widget is always black, but if this widget is set to full screen and then cut to small screen, it will be normal.

dzx815, 2025-06-27:
已解决

closed 2025-06-27 by dzx815