Serial-Studio/Serial-Studio #317: Can connect when trial is finished

maximeg, 2025-08-05:
Hello,

After a  bit of time off, I tried to use Serial Studio and I don't have a "Connect" button now that the trial is finished.
I reviewed my project in the project editor and I don't use any pro feature.

Version is 3.1.7

[image]

maximeg, 2025-08-05:
Same thing with last dev build (https://github.com/Serial-Studio/Serial-Studio/actions/runs/16715120614)

alex-spataru, 2025-08-06:
#### For added context on why builds now include a 14‑day Pro trial before locking down:

The [LICENSE.md](https://github.com/Serial-Studio/Serial-Studio/blob/master/LICENSE.md) explains the terms in detail. In short, building Serial Studio yourself produces a GPLv3 build with some Pro features excluded. These are controlled via CMake flags, enabling them requires a valid license key and activation ID, so you cannot “accidentally” compile a Pro version. 

The full source, including the activation mechanism, is public because I believe in transparent, auditable software. The GPLv3 build includes the same functionality as older versions of Serial Studio, plus improvements like the new dashboard layout, a friendlier Project Editor and performance/stability upgrades.

#### On the “why can’t I use this for free” topic (from my point of view as a solo dev):

I recommend watching [this video](https://www.youtube.com/watch?v=vzYqxo13I1U&t=2646s). In summary:

> People who share their source code do not owe you anything. If I give you the source of one of my projects, I’ve given you the source and the freedom to use it. You don’t get an upgrade, you don’t get support, you don’t get bug fixes, you don’t get maintenance, you don’t get dibs on my email, or my time, or any of that kind of thing. If you need support, pay for it.

I want Serial Studio to remain sustainable so I can keep improving it for everyone. In the past, it’s been used in commercial settings without contributions back, sometimes paired with a rush of conflicting feature requests within just a few days. I’ve also been blamed for issues that could have been solved by taking 5 minutes to read the wiki, the README, or the code. My goal is to maintain a healthier development cycle, one where users who benefit from the project also help it grow, whether through funding, ideas, code, or community support.

I believe the current pricing is fair for both individuals and businesses, and the dual‑license model is designed for people who genuinely value open source. Paying users directly improve the software’s stability and capabilities. GPLv3 users can still add value by contributing code, ideas, or fixes.

The “price” of getting a GPLv3 build is stepping out of your comfort zone to learn how Serial Studio works internally and how to improve it. I have no interest in crippling the GPLv3 build to force upgrades, but building it yourself shows you care enough to invest 30 minutes installing Qt and compiling Serial Studio. If you do that, you’ve earned a voice in the project, even without contributing financially.

#### Regarding the difference between Pro and GPLv3:

I understand that right now many users may not see a big functional difference between Pro and GPLv3, which can cause confusion. Over time, Pro will focus on features for businesses and professional use, while GPLv3 will keep the feature set many users already rely on. The value of Pro right now is that it keeps this project alive instead of becoming abandonware, and your feedback will guide how it evolves.

As always, I’m open to suggestions on how to balance community needs with keeping Serial Studio viable long‑term. I’m grateful for the many great conversations and collaborations I’ve had over the years, and I want to keep those kinds of users close while filtering out the “I only use open source to avoid paying” crowd.

closed 2025-08-11 by alex-spataru